elixir-circuits/circuits_uart #3: Not working on OSX: argument error     :erlang.port_close(#Port<#####>)

jisaacstone, 2016-05-21:
Not having success on OSX

```
iex(1)> Nerves.UART.enumerate
** (ArgumentError) argument error
    :erlang.port_close(#Port<0.182363>)
    lib/uart/enumerator.ex:17: Nerves.UART.Enumerator.enumerate/0
```

```
Interactive Elixir (1.2.4) - press Ctrl+C to exit (type h() ENTER for help)
iex(1)> {:ok, pid} = Nerves.UART.start_link
{:ok, #PID<0.13578.0>}
Going to terminate: {:function_clause, [{Nerves.UART, :handle_info, [{#Port<0.20777>, {:exit_status, 1}}, %{controlling_process: nil, name: nil, port: #Port<0.20777>}], [file: 'lib/nerves_uart.ex', line: 286]}, {:gen_server, :try_dispatch, 4, [file: 'gen_server.erl', line: 615]}, {:gen_server, :handle_msg, 5, [file: 'gen_server.erl', line: 681]}, {:proc_lib, :init_p_do_apply, 3, [file: 'proc_lib.erl', line: 240]}]}
** (EXIT from #PID<0.385.0>) an exception was raised:
    ** (ArgumentError) argument error
        :erlang.port_close(#Port<0.20777>)
        (stdlib) gen_server.erl:643: :gen_server.try_terminate/3
        (stdlib) gen_server.erl:809: :gen_server.terminate/7
        (stdlib) proc_lib.erl:240: :proc_lib.init_p_do_apply/3

Interactive Elixir (1.2.4) - press Ctrl+C to exit (type h() ENTER for help)
iex(2)>
20:35:03.103 [error] GenServer #PID<0.13578.0> terminating
** (ArgumentError) argument error
    :erlang.port_close(#Port<0.20777>)
    (stdlib) gen_server.erl:643: :gen_server.try_terminate/3
    (stdlib) gen_server.erl:809: :gen_server.terminate/7
    (stdlib) proc_lib.erl:240: :proc_lib.init_p_do_apply/3
Last message: {#Port<0.20777>, {:exit_status, 1}}
State: %{controlling_process: nil, name: nil, port: #Port<0.20777>}
```

seems the port opened [here](https://github.com/nerves-project/nerves_uart/blob/master/lib/uart/enumerator.ex#L6) is in a closed state before it is closed [here](https://github.com/nerves-project/nerves_uart/blob/master/lib/uart/enumerator.ex#L17)

But I cannot figure out how or why ....

fhunleth, 2016-05-21:
Ugh! Could you try running `priv/nerves_uart` from the command line and see if it provides any hints? If you run it without arguments, it should look like it is hanging since it's waiting for input. If you run it and pass `enumerate` as the first argument, it should print out something that vaguely looks like the serial ports attached to your system. My hope is that it says something else.

mobileoverlord, 2016-05-21:
Resolved. This was occurring because the port was cross-compiled and was being loaded on the host.

closed 2016-05-21 by mobileoverlord