elixir-circuits/circuits_uart #10: Can not install on Windows 10. "Dependency not locked"

Gyllsdorff, 2016-10-11:
```
C:\workspace\elixpos>mix deps.get
Running dependency resolution
* Getting nerves_uart (Hex package)
  Checking package (https://repo.hex.pm/tarballs/nerves_uart-0.1.1.tar)
  Fetched package
* Getting elixir_make (Hex package)
  Checking package (https://repo.hex.pm/tarballs/elixir_make-0.3.0.tar)
  Fetched package

C:\workspace\elixpos>mix compile
==> elixir_make
Compiling 1 file (.ex)
Generated elixir_make app
==> nerves_uart
Unchecked dependencies for environment prod:
* elixir_make (Hex package)
  the dependency is not locked (run "mix deps.get" to generate "mix.lock" file)
could not compile dependency :nerves_uart, "mix compile" failed. You can recompile this dependency with "mix deps.compile nerves_uart", update it with "mix deps.update nerves_uart" or clean it with "mix deps.clean nerves_uart"
==> elixpos
** (Mix) Can't continue due to errors on dependencies

C:\workspace\elixpos>mix deps.get
Running dependency resolution
All dependencies up to date

C:\workspace\elixpos>mix deps.compile nerves_uart
==> nerves_uart
Unchecked dependencies for environment prod:
* elixir_make (Hex package)
  the dependency is not locked (run "mix deps.get" to generate "mix.lock" file)
could not compile dependency :nerves_uart, "mix compile" failed. You can recompile this dependency with "mix deps.compile nerves_uart", update it with "mix deps.update nerves_uart" or clean it with "mix deps.clean nerves_uart"
==> elixpos
** (Mix) Can't continue due to errors on dependencies

C:\workspace\elixpos>dir mix.lock
 Volume in drive C is OS
 Volume Serial Number is 7EF8-261C

 Directory of C:\workspace\elixpos

11/10/2016  18:46               557 mix.lock
               1 File(s)            557 bytes
               0 Dir(s)  35,903,332,352 bytes free
```

Key | Value
------------ | -------------
OS: | Windows 10, sight.
Elixir: | 1.3.2
Erlang: | 19.1
Mix env: | dev
Dependencies in mix.env | `{:nerves_uart, "~> 0.1.1"}`

I can compile and build other elixir projects without a problem.

fhunleth, 2016-10-11:
I just wanted to post some thoughts from asking around on this issue. None of us currently have a Windows 10 machine accessible, but we're thinking that this is either a mix or elixir_make issue. We're not sure why you're seeing it and we haven't yet, though. I do see that you're running slighly newer Erlang and Elixir than I am, but I don't know why that would cause the problem. If you get impatient waiting for us, I might try asking around on the elixir-lang slack in case other Windows users have seen this.

Gyllsdorff, 2016-10-11:
There is no rush, I have access to other Linux machines.  :)

I can try to install older Elixir/Erlang combinations tomorrow to test if you think that could help? Is there any particular version of Erlang/Elixir/Mix that you know have worked for other people? 

Frankly, it almost feels like this is a bug in hex/mix or something since the actual error message is from mix. I can't figure out why it complains about `the dependency is not locked (run "mix deps.get" to generate "mix.lock" file)`. 

When I tried using `{:nerves_uart, "0.1.0"}` I at least got a real `mingw32-make not found` error.

fhunleth, 2016-10-11:
I'm using Erlang 19.0 and Elixir 1.3.1 on Windows 10. Both were installed via Chocolatey.

jmerriweather, 2016-10-12:
I'm also on Windows 10 and for me this went away after doing the following from an admin command prompt:
```
choco install mingw
```

Gyllsdorff, 2016-10-12:
> I'm also on Windows 10 and for me this went away after doing the following from an admin command prompt:
> 
>```
> choco install mingw
> ```

That solved it. It seems like `On Windows, if you're obtaining nerves_uart from hex.pm, you'll get a precompiled version of the C code. This will be used if the proper build tools aren't detected.` is not always correct.

closed 2016-10-12 by Gyllsdorff

fhunleth, 2016-10-12:
@Gyllsdorff or @jmerriweather Thanks for following up with the solution! If either of you have hints to why the precompiled binary didn't work, could you open up a new issue? If nothing else, the error message you get when it doesn't work is surprisingly unhelpful and I'd like to see if I can get it to figure out what's happening.

Gyllsdorff, 2016-10-12:
I can probably take a look at it this weekend.

One question: if I did not have mingw installed I was supposed to get a `"Copying prebuilt port binary"` in the cmd window when it tried to copy the pre-compiles files, right? I will try adding a few debug statements and see which path the code follows in `nerves_uart/mix.env`.

fhunleth, 2016-10-12:
Right. That was the idea. Thanks!