elixir-circuits/circuits_uart #25: Error when running from an escript

pallix, 2017-11-09:
When running the library as an escript, an error is raised:

```
** (EXIT from #PID<0.77.0>) an exception was raised:
    ** (ArgumentError) argument error
        :erlang.++({:error, :bad_name}, '/nerves_uart')
        (nerves_uart) lib/nerves_uart.ex:284: Nerves.UART.init/1
        (stdlib) gen_server.erl:365: :gen_server.init_it/2
        (stdlib) gen_server.erl:333: :gen_server.init_it/6
        (stdlib) proc_lib.erl:247: :proc_lib.init_p_do_apply/3

```

mobileoverlord, 2017-11-09:
nerves_uart includes c code that is required to interface with the serial ports. This code is compiled and included in the priv dir. Because of this, nerves_uart is not able to be included in and called form an escript. This is due to limitations with escripts. See this note from the elixir docs regarding escripts and priv dir.
https://github.com/elixir-lang/elixir/blob/edd7f24a0ecca668b330154ac7d4ca50e0be8dff/lib/mix/lib/mix/tasks/escript.build.ex#L35-L36

pallix, 2017-11-09:
I see. Thank you for the explanation!

closed 2017-11-09 by mobileoverlord