elixir-circuits/circuits_uart #29: Fix Dialyzer errors

ejpcmac, 2017-12-19:
I am running into a Dialyzer error when using `Nerves.UART.read/2` from v0.1.2, as the `@spec` is not correct.

Searching through the repo, I see it has been fixed more than six months ago in #20, but it is not yet released. Is there a plan to do a bugfix release anytime soon? It is sad to get a cascade of errors in Atom ide-elixir for this.

fhunleth, 2017-12-19:
Thanks for reminding me! I haven't been using `nerves_uart` recently and it fell off my todo list. I just published a new version on hex with #20 and a couple other minor fixes. Hope it works for you now.

closed 2017-12-19 by fhunleth

ejpcmac, 2017-12-19:
No problem. Thank you for writing this library :) It works perfectly!