elixir-circuits/circuits_uart #38: Use pid instead of serial port name

tallakt, 2018-03-21:
Hi. When reading serial data through messages (active: true), the data received is in the form ‘{:nerves_uart, port_name, data}’. Because og this, I need to maintain a map of name to pids, as pids are what I can use to send messages back to the receiving serial port. I would have expected and preferred the messages to be in the format:  ‘{:nerves_uart, pid, data}’. This could be implemented in a backeard compatible way by having an argument to ‘open’ such as: ‘open(pid, name, active: :format2)

fhunleth, 2018-03-22:
I agree that's annoying. Please feel free to send a PR with this change.

tallakt, 2018-03-28:
https://github.com/nerves-project/nerves_uart/pull/39

closed 2018-03-28 by tallakt