elixir-circuits/circuits_uart #40: Remove dependency of hardware to run tests with socat

tallakt, 2018-03-28:
Instead of using a cable, some tests could also run using a null modem simulation using `socat`. Perhaps even this should run by default, and could run in Travis. I tried this quickly and mosts tests run with this quick setup:

In one shell:

```
$ socat -d -d pty,raw,echo=0 pty,raw,echo=0
2018/03/28 14:12:14 socat[97930] N PTY is /dev/ttys002
2018/03/28 14:12:14 socat[97930] N PTY is /dev/ttys003
2018/03/28 14:12:14 socat[97930] N starting data transfer loop with FDs [5,5] and [7,7]
```

In another shell:

```
$ export NERVES_UART_PORT2=ttys003
$ export NERVES_UART_PORT1=ttys002
$ mix test
```

fhunleth, 2018-03-28:
`nerves_uart` uses [tty0tty](https://github.com/freemed/tty0tty) on Travis and Windows has COM0COM. Those both support hardware line emulation which is something I thought was missing from socat. Perhaps this is something for OSX which doesn't have a serial port emulator?

tallakt, 2018-03-28:
Sorry did not read the complete docs. It might be an option for OSX, but no time to test that at the moment.

closed 2018-03-28 by tallakt