elixir-circuits/circuits_uart #41: Program exits when framing is incorrect.

acrogenesis, 2018-04-04:
If we set the option for framing with `\n` but send no new line we get this error:
```
** (exit) exited in: GenServer.call(#PID<0.166.0>, {:read, 3000}, 3100)
    ** (EXIT) time out
    (elixir) lib/gen_server.ex:834: GenServer.call/3
    (hello_uart) lib/hello_uart.ex:24: HelloUart.uart_loop/1
```

The example can be found in: https://github.com/DisruptiveAngels/nerves-uart-example

fhunleth, 2018-04-04:
Thanks for the error report and example. What OS and hardware are you using?

acrogenesis, 2018-04-04:
```
macOS 10.13.4
Erlang/OTP 20 [erts-9.3] [source] [64-bit] [smp:8:8] [ds:8:8:10] [async-threads:10] [hipe] [kernel-
poll:false]
Elixir (1.6.4)
```
I think what's happening is that it's waiting for the `\n` but because it wasn't sent it just timeouts. Perhaps it should just send whats on the buffer or a timeout error.

acrogenesis, 2018-04-04:
For testing this we run the example on my Mac and on another pc connected through USB/uart and on the pc we run Arduino to easily send messages without line ending, `\n`, and `\r\n`

acrogenesis, 2018-04-10:
So I was debugging the error more and noticed I was using the version 1.0.1 when I updated it started working as expected

closed 2018-04-10 by acrogenesis