elixir-circuits/circuits_uart #44: rpi0 Nerves.UART.enumerate times out and gets argument error

chiragtoor, 2018-05-22:
## Setup

  - Version: latest
  - OS: OSX
  - Platform: Raspberry Pi 0
  
## Expected Behavior
`Nerves.UART.enumerate` should return a map of available ports with info on each.

## Actual Behavior
The receive block in `Nerves.UART.Enumerator` times out and gets a argument error on the call to `Port.close()` (line 21).

## Steps to Reproduce the Problem
1. Running the hello_network nerves_example repo, ssh to a IEx console
2. From IEx run `Nerves.UART.enumerate`, after 5 seconds it should error out

mattludwigs, 2018-05-22:
I ran through this issue on my RPi 0 and I did not experience the behavior from the above the issue.

Is there any more information you can provide so maybe I can try to reproduce?

For example, I was connecting with Wifi, are you connecting using `eth0` or `usb0`?

Any other details you can think of would be great, so we can reproduce your set up. Thank you!

chiragtoor, 2018-05-24:
I am connecting via WiFI, I noticed a error when I try to call `Nerves.UART.start_link()`, these are my outputs:

```
iex(1)> Nerves.UART.enumerate
iex(2)> 
** (ArgumentError) argument error
    :erlang.port_close(#Port<0.3383>)
    (nerves_uart) lib/uart/enumerator.ex:21: Nerves.UART.Enumerator.enumerate/0
iex(1)> 
nil
iex(2)> 
nil
iex(3)> {:ok, pid} = Nerves.UART.start_link()
{:ok, #PID<0.427.0>}
** (EXIT from #PID<0.422.0>) shell process exited with reason: an exception was raised:
    ** (ArgumentError) argument error
        :erlang.port_close(#Port<0.3384>)
        (stdlib) gen_server.erl:648: :gen_server.try_terminate/3
        (stdlib) gen_server.erl:833: :gen_server.terminate/10
        (stdlib) proc_lib.erl:247: :proc_lib.init_p_do_apply/3
```

It's odd I see a response `{:ok, #PID<0.427.0>}` and then right afterwards it errors out.

My entire config is:
```
config :logger, level: :debug

config :hello_network, interface: :wlan0

key_mgmt = "WPA-PSK"

config :nerves_network, :default,
  wlan0: [
    ssid: "[NETWORK_USERNAME]",
    psk: "[NETWORK_PASSWORD]",
    key_mgmt: String.to_atom(key_mgmt)
  ],
  eth0: [
    ipv4_address_method: :dhcp
  ]

config :nerves, :firmware,
  rootfs_overlay: "rootfs_overlay"

config :shoehorn,
  init: [:nerves_runtime, :nerves_init_gadget],
  app: Mix.Project.config()[:app]

config :nerves_leds, names: [green: "led0"]

config :nerves_firmware_ssh,
  authorized_keys: [
    File.read!(Path.join(System.user_home!(), ".ssh/id_rsa.pub"))
  ]

config :nerves_init_gadget,
  ssh_console_port: 22,
  node_name: :hello_network,
  mdns_domain: "hello_network.local"
```

chiragtoor, 2018-05-24:
This has worked on a rpi3 for me without any issues

chiragtoor, 2018-05-24:
I think I see what it is, I was originally working with a rpi3 and had things working, I did this when I switched to a rpi0:

`MIX_TARGET=rpi0 mix do deps.get, compile, firmware, firmware.burn`

I figured that would get the deps for the new target and work properly, but just now I tried first cleaning out the deps: `mix deps.clean --all` and then redid the above command, now it seems to be working.

I'm not sure exactly what happened, some deps were left over that were compiled for rpi3 possibly? I thought `MIX_TARGET=rpi0 mix do deps.get, compile` would have all deps ready for a rpi0.

chiragtoor, 2018-05-24:
It's been working fine now after I cleaned the deps, able to receive the data I was trying to receive. I'll close this issue, thank you everyone for taking a look.

closed 2018-05-24 by chiragtoor