elixir-circuits/circuits_uart #45: Building on Alpine Linux

entone, 2018-08-24:
I worked through getting `nerves_uart` building on x86_64 Alpine Linux today and wanted to share my setup.

This is part of a multi-part build in Docker so I'll just share my Dockerfile
[Dockerfile.txt](https://github.com/nerves-project/nerves_uart/files/2319546/Dockerfile.txt)

The relevant parts are the apk packages that are installed.

`RUN apk add --update git erlang-dev build-base make gcc abuild binutils cmake linux-headers`

Hopefully this can help other people looking to do the same.

entone, 2018-08-24:
> cmake is needed?

Probably not, I was trying some things.

entone, 2018-08-27:
Ok, ran a few more tests, and this appears to be the minimum packages required.

`RUN apk add --update git erlang-dev build-base linux-headers`

fhunleth, 2019-04-03:
Resolving old issue that was actually resolved.

closed 2019-04-03 by fhunleth