elixir-circuits/circuits_uart #54: Discussion: Move this to elixir-circuits umbrella?

jjcarstens, 2018-11-25:
With all the new `[elixir-circuits](https://github.com/elixir-circuits)` work going on for GPIO, I2c, and SPI, would it make sense to move `nerves_uart` to that umbrella instead as `circuits_uart`?

It seems to be a common misconception that one would need nerves to use this and that might help separate the distinction from nerves.

But on the flip side, it might not be technically a “circuit” since UART is not dependent on pin headers and is focused on being cross platform library.

fhunleth, 2018-11-26:
Yes, totally agree. Our plan is to move it. 

I was going to do it after we released `circuits_gpio`. You're the n'th person to ask me about this, though, so let me see if I can just get it done.

jjcarstens, 2018-11-26:
I’d be happy to help and get it started if you can just create the repo in elixir circuits

fhunleth, 2018-11-26:
That would be awesome! Several things just popped up at work and now I'm not sure if I can get to this today or tomorrow, but the first step is just renaming everything to `circuits_uart`. I assume that it's a simple search and replace, but I'm sure that there will be a few places that need extra scrutiny. If you want to take a pass at that in a fork, then I'll pull your fork when I set up the new repo and update the CI settings. That would save me quite a bit of time and probably get things most of the way there.

fhunleth, 2018-11-26:
You're totally making my day! Thanks!

I really hope to get a chance to take the next pass tonight.

fhunleth, 2018-11-27:
Here's where I'm at: https://github.com/elixir-circuits/circuits_uart

I think that it's really close. Unit tests pass and the docs look good at a glance. Need to button it up in a release and post about it - probably later this week.

fhunleth, 2019-01-02:
Moved

closed 2019-01-02 by fhunleth