elixir-circuits/circuits_uart #70: Latency

robinhilliard, 2019-11-08:
## Setup

  - Version: latest (with PR fork)
  - OS: OSX
  - Platform: MacBook Pro
  
## Expected Behavior

No significant latency

## Actual Behavior

Latency up to a second with modems set to 20ms window to minimise buffering

## Steps to Reproduce the Problem

I have a pair of radio modems at close to point blank range set to 57600 baud sending a short < 100 byte message. The modems have a firmware maximum window size config of 20ms which is the setting normally used to minimise latency. I minimised the number of other messages, and the same software setup running over WiFi/UDP has no noticable delay. However when I use the modems with one of them via circuits UART I get a latency delay of about a second. I have set the uart connection to active.

I know there are lots of variables in play here, but I was wondering if there are any configuration settings you would use to minimise latency of short messages when using this library?

Thanks,
Robin

fhunleth, 2019-11-08:
My use of Circuits UART isn't performance critical and I know that Circuits.UART performance could be improved. However, I wasn't expecting issues on an MBP at 57600. 

Just so I understand right, the firmware at the other end times out since it doesn't receive a message in 20 ms. At 57600 baud, a 100 byte message takes about 17 ms to send. If it's delayed by 3 ms, then it times out?

robinhilliard, 2019-11-08:
Hi Frank,

Sorry the window size just means that it sends what it has in it’s buffers every 20ms. I can see the timeout arguments in the C code and mention of polling but don’t understand what it’s doing, and wondered if perhaps there was a polling interval or other setting in there that could be tweaked.

Cheers
Robin

fhunleth, 2019-11-08:
Does the firmware work with any other software on your Mac? For example, does it work with Picocom? It's sounding like it's not a Circuits.UART issue, but a general issue that writing bytes to the serial port doesn't work well.

Is the C code that you're referring to posted anywhere?