elixir-circuits/circuits_uart #81: Cannot set parity to :mark

robinhilliard, 2020-04-08:
## Setup

  - Version: 1.4.1
  - OS: OSX 10.14.6
  - Platform: MacBook
  - Silicon Labs CP210x v5 USB to UART Driver
  
## Expected Behavior

The code works with parity set to `:ignore` but according to the documentation and code comments I should be able to set parity to `:mark`.
```
{:ok, pid} = Circuits.UART.start_link()
:ok = Circuits.UART.open(pid, "/dev/tty.SLAB_USBtoUART", speed: 9600, parity: :mark, data_bits: 8, stop_bits: 1, active: true)
```
## Actual Behavior

open returns `{:error, :einval}`.

## Steps to Reproduce the Problem

Above two lines of code.

fhunleth, 2020-04-08:
I don't think `:mark` parity is supported on OSX. The way that it's implemented for Linux is by using the `CMSPAR` option to `tcsetattr`. I'm pretty sure that `CMSPAR` doesn't exist on Darwin. 

I haven't looked in the past year or so, so maybe Apple added it? If you know of an OSX way of enabling it or have any other hints, please let me know.

robinhilliard, 2020-04-08:
Right, thanks for explaining and no sadly I have no additional information for you. It looks like it can be dodgy using mark and space even on Linux.

closed 2020-04-08 by robinhilliard