elixir-circuits/circuits_uart #82: ENOTTY error setting custom speed on linux

tomboland, 2020-04-27:
## Setup
circuits_uart 1.4.1
OS is Manjaro Linux, Kernel 5.6.6
I7 Laptop
Communicating with a USB serial cable to a DSP amplifier (Mac Audio Reference 2.1 DSP)
The software I'm writing is here: [https://github.com/tomboland/mac_audio_reference_2.1_dsp](https://github.com/tomboland/mac_audio_reference_2.1_dsp)

This function is the first thing I'm trying to get working: [https://github.com/tomboland/mac_audio_reference_2.1_dsp/blob/a10b572811bb51540c745f51f78bba553963fd00/lib/dsp_uart.ex#L56](https://github.com/tomboland/mac_audio_reference_2.1_dsp/blob/a10b572811bb51540c745f51f78bba553963fd00/lib/dsp_uart.ex#L56)

I'm invoking it in this test here: [https://github.com/tomboland/mac_audio_reference_2.1_dsp/blob/a10b572811bb51540c745f51f78bba553963fd00/test/dsp_uart_test.exs#L68](https://github.com/tomboland/mac_audio_reference_2.1_dsp/blob/a10b572811bb51540c745f51f78bba553963fd00/test/dsp_uart_test.exs#L68)

This is my first time using Elixir, and I've found it a pleasure to use, but you may find some stuff unidiomatic!  I'm up for trying to submit a PR, I'm not bad with C, but I'll probably find it difficult to dive in and get this sorted.  I'm hoping someone may have an instinct on it from the behaviour I'm describing! Thanks for any help, and please let me know if I can give you any more information, or be of any help at all (with guidance)! :)
  

## Expected Behavior
I think I should be able to set a custom BAUD rate of 192000 or 200000. These speeds both work in an example python script I'm pasting below, which uses pyserial.  I've tried with various other more conventional speeds, like 115200, 230400 and others, and I fail to communicate with my DSP using those speeds in python, but 192000 and 200000 work OK.  The original software uses 192000.  I think the problem is with the ioctl being used, but I must admit, it's not something I'm confident to make assertions about.  The expected behaviour can be seen in the strace output from the python script that I'll put below.  I'm also pasting strace output from my elixir program.

## Actual Behavior
I get an ENOTTY error and cannot use the serial port at all when I configure a speed of 192000 or 200000.    

## Steps to Reproduce the Problem
I don't think you'd need my hardware to reproduce this, and actually, if I use tty0tty and try and connect to /dev/tnt0, I also get the ENOTTY error.  Please see the strace output below, which is talking to the real hardware.  You can see that the python script (where I've basically copypasta'd bits of the original code I'm rewriting) is getting the response I expect from the serialised message.  Basically, if I send this binary: <<85, 2, 3, 7, 10>>, I should get back <<0xff>>

## Python script ##
```python
from serial import Serial, SerialException
import ctypes
import array

def send_message(ser, message):
    if len(message) == 0:
        return
    ln = len(message) // 2 + 1
    buf = array.array('B', [85, ln, 3])
    sum = 3
    for i in range(len(message) // 2):
        m = int(message[i * 2:i * 2 + 2], 16)
        sum = sum + m
        buf.append(m)

    buf.append(ctypes.c_ubyte(sum).value)
    print("DEBUG:", buf, buf.tostring())
    ser.write(buf.tostring())

if __name__ == "__main__":
    ser = Serial()
    ser.baudrate = 192000
    ser.timeout = 2
    ser.port = '/dev/ttyUSB0'
    try: ser.open()
    except SerialException: pass
    send_message(ser, b'07')
    response = ser.read()
    print(response)
    send_message(ser, b'08')
    response = ser.read()
```

## strace output from above srcipt (stdout/DEBUG messages show the expected response being recieved.  I've limited the strace to -e trace=ioctl ##

```
ioctl(3, TCGETS, 0x7fffedf6bff0)        = -1 ENOTTY (Inappropriate ioctl for device)
ioctl(3, TCGETS, 0x7fffedf6bff0)        = -1 ENOTTY (Inappropriate ioctl for device)
ioctl(3, TCGETS, 0x7fffedf6b0a0)        = -1 ENOTTY (Inappropriate ioctl for device)
ioctl(3, TCGETS, 0x7fffedf6a7e0)        = -1 ENOTTY (Inappropriate ioctl for device)
ioctl(3, TCGETS, 0x7fffedf6a7e0)        = -1 ENOTTY (Inappropriate ioctl for device)
ioctl(3, TCGETS, 0x7fffedf6c010)        = -1 ENOTTY (Inappropriate ioctl for device)
ioctl(3, TCGETS, 0x7fffedf6c010)        = -1 ENOTTY (Inappropriate ioctl for device)
ioctl(3, TCGETS, 0x7fffedf6c020)        = -1 ENOTTY (Inappropriate ioctl for device)
ioctl(3, TCGETS, 0x7fffedf6c020)        = -1 ENOTTY (Inappropriate ioctl for device)
[... 203 lines cut ...]
[pid 833274] +++ exited with 1 +++
[pid 833273] +++ exited with 1 +++
[pid 833272] +++ exited with 1 +++
[pid 833271] +++ exited with 1 +++
[pid 833270] +++ exited with 1 +++
[pid 833269] +++ exited with 1 +++
[pid 833268] +++ exited with 1 +++
[pid 833267] +++ exited with 1 +++
[pid 833266] +++ exited with 1 +++
[pid 833265] +++ exited with 1 +++
[pid 833257] +++ exited with 1 +++
[pid 833264] +++ exited with 1 +++
[pid 833263] +++ exited with 1 +++
[pid 833262] +++ exited with 1 +++
[pid 833261] +++ exited with 1 +++
[pid 833260] +++ exited with 1 +++
[pid 833259] +++ exited with 1 +++
[pid 833258] +++ exited with 1 +++
[pid 833256] +++ exited with 1 +++
[pid 833253] +++ exited with 1 +++
[pid 833252] +++ exited with 1 +++
[pid 833279] +++ exited with 1 +++
[pid 833240] +++ exited with 1 +++
+++ exited with 0 +++
```

fhunleth, 2020-04-27:
Oh, I didn't know about `TCGETS2`!

This code looks interesting: https://github.com/pyserial/pyserial/blob/master/serial/serialposix.py#L148-L162

It looks like there's a better way to do this: https://github.com/elixir-circuits/circuits_uart/blob/master/src/uart_comm_unix.c#L183-L194

Apologies, no one ever sends me `straces` and they're so helpful that I skimmed over almost everything else you posted when I saw TCGETS2. When I get more time, I'll read this more carefully. Thanks for the report!!!

tomboland, 2020-04-27:
Hi, thanks for the amazingly quick response!  You may see I've just submitted a PR which uses those newer ioctls. I added a simple test, and all of the other tests are still passing (on my machine, fingers crossed on Travis!). Please let me know what you think!

tomboland, 2020-04-27:
And actually, I can talk to my DSP now! It works! :)

fhunleth, 2020-05-04:
Fixed in f17ca766c11ad3f1a4859b38067df6e9929c84c1.

closed 2020-05-04 by fhunleth