elixir-circuits/circuits_uart #88: UART.read timeout clarifications

mattias01, 2020-11-04:
When a timeout happens upon calling UART.read, it returns `{:ok, <<>>}`. 
See code:
https://github.com/elixir-circuits/circuits_uart/blob/6ffd4393afa5aa48a97271fc747c96d0598c536a/lib/circuits_uart.ex#L417-L418

This is not intuitive to me. I expect to get something like `{:ok, :timeout}` or `{:error, :timeout}`.

I suggest implementing one of three options below to improve this:

1. Improve the documentation, so that it is explicit in how to detect a timeout.
2. Instead or returning `{:ok, <<>>}` on a timeout, return `{:ok, :timeout}` or `{:error, :timeout}`. This is a breaking change.
3. Add a third optional parameter with what to return if a timeout happens. Something like: `read(pid, timeout \\ 5000, result_on_timeout \\ <<>>)`. I don't think this is considered a breaking change.

I'm willing to submit a PR if we can come to a conclusion on what to do.

fhunleth, 2020-11-05:
Thanks for asking about the timeout. The API is mostly frozen now, so my preference would be to add a note to the documentation.

mattias01, 2020-11-06:
Ok! That's alright. I will add a PR for that in a not too distant future.

closed 2020-11-06 by mattias01