elixir-circuits/circuits_uart #90: Mingw 10.2 compilation is failing

bobkocisko, 2021-01-19:
## Setup

  - Version: 1.4.2
  - OS: Windows 10
  
## Expected Behavior
Successful compilation

## Actual Behavior
Compilation fails at link time:
```
c:/programdata/chocolatey/lib/mingw/tools/install/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: c:/dev/bas/bas_server/_build/dev/lib/circuits_uart/obj/util.o:util.c:(.text+0x40): undefined reference to `clock_gettime'
collect2.exe: error: ld returned 1 exit status
mingw32-make: *** [src/Makefile:111: c:/dev/bas/bas_server/_build/dev/lib/circuits_uart/priv/circuits_uart.exe] Error 1
could not compile dependency :circuits_uart, "mix compile" failed. You can recompile this dependency with "mix deps.compile circuits_uart", update it with "mix deps.update circuits_uart" or clean it with "mix deps.clean circuits_uart"
** (Mix) Could not compile with "mingw32-make" (exit status: 2).
You may need to install mingw-w64 and make sure that it is in your PATH. Test this by
running `gcc --version` on the command line.
```

## Steps to Reproduce the Problem

Build on Windows

fhunleth, 2021-01-22:
Sorry for the delay in looking at this. I'm not actively using Windows so it might take me a bit to debug. If the issue is that `clock_gettime` was removed from mingw, I would assume there's a workaround. If you find something and submit a PR, I'll make time to give it a quick test and make an official release.

bobkocisko, 2021-01-22:
Thanks!  For now I've just pinned mingw at 8.1.0 in chocolatey and all seems to be well.