elixir-circuits/circuits_uart #99: Issues with Fona 808 on Raspberry Pi Zero W

arduino-man, 2021-04-25:
## Setup

  - Version: Circuits UART 1.4.2
  - OS: Elixir 1.11.2  -  Nerves 1.7.5 
  - Platform: Raspberry Pi Zero W
  
## Expected Behavior
Talk to a Fona 808 device using Circuits UART on a Raspberry Pi Zero W.

## Actual Behavior
Cannot talk to Fona 808 when using a Raspberry Pi Zero W, the responses from the device are either empty or "Unknown Command" or a bunch of "\b\b\b\b\b\b\b\b\b\b\b" characters.

## Steps to Reproduce the Problem
I have a Fona 808 device which I am able to talk to via UART using a USB to TTL adapter and Circuits UART on my laptop with IEX but cannot do the same when using Circuits UART on a Raspberry Pi Zero W and Nerves.

SAMPLE OUTPUT:

iex(5)> Circuits.UART.read(pid, 1000)
{:ok, ""}

iex(5)> Circuits.UART.read(pid, 1000)
{:ok,
 "d\r --> > Unknown command\rUnknown command\r --> n command\rUnknown command\r"}

iex(5)> Circuits.UART.read(pid, 1000)
{:ok,
 {:partial,
  " \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b \b "}}

[image]
[image]

fhunleth, 2021-04-25:
Can you confirm that the baud rate is set correctly?

fhunleth, 2021-04-25:
My second idea is to add this to your config:

```
  config :nerves, :erlinit,
    ctty: "/dev/null"
```

I just realized that you're on a Pi Zero and the serial console should be running on the same UART. I.e., it's sending IEx prompts to the Fona 808 and that's probably not good at all.

arduino-man, 2021-04-25:
I can confirm that the baud rate is set correctly (115200).
Is there anything I can do to prevent the issue with IEX commands being sent to the Fona via UART?  Thanks for your prompt response

arduino-man, 2021-04-25:
Oh sorry, I just realized your suggestion was part of your comment, making that change now and will report right back

arduino-man, 2021-04-25:
I added the new configuration line redirecting output to /dev/null but no success.

My config file can be seen here in case I did anything wrong.  https://pastebin.com/KLEPmEYD

arduino-man, 2021-04-25:
I also found this forum entry where someone encountered what seems to be the exact same issue. The issue seems to have never been resolved unfortunately.  https://elixirforum.com/t/circuiits-uart-problems-in-rpi0/33710/7

arduino-man, 2021-04-25:
Doing 

config :nerves, :erlinit, ctty: "ttyS0"

Seems to have resolved the issue or at least dramatically improved it. I will stress test and report back in an hour or so.   

Thanks!

fhunleth, 2021-04-25:
Weird. Another thing to try is to turn off kernel messages being output over the UART. I'd manually try this out first by putting the MicroSD card in your computer and editing the `cmdline.txt` file. The line at the end says this:

```
root=/dev/mmcblk0p2 rootwait console=serial0,115200 quiet
```

Delete the console part so it's like this:

```
root=/dev/mmcblk0p2 rootwait quiet
```

arduino-man, 2021-04-25:
Adding: 

config :nerves, :erlinit, ctty: "ttyS0"

completely fixed the issue! Thanks a lot for pointing me in the right direction.

closed 2021-04-25 by fhunleth