elixir-circuits/circuits_uart #113: Possible option to set timeout on Open ?

CharlesOkwuagwu, 2021-10-30:
## Setup

  - Version:  1.4.3
  - OS: Windows
  - Platform: Windows 10
  
`circuits_uart.open` command has no way to configure a `timeout`, the default 4000ms always applies, and the error below might be avoidable if we could **configure** a longer or **flexible** timeout say.
```
** (stop) :port_timed_out
    (circuits_uart 1.4.3) lib/circuits_uart.ex:563: Circuits.UART.call_port/4
    (circuits_uart 1.4.3) lib/circuits_uart.ex:359: Circuits.UART.handle_call/3
    (stdlib 3.16.1) gen_server.erl:721: :gen_server.try_handle_call/4
    (stdlib 3.16.1) gen_server.erl:750: :gen_server.handle_msg/6
    (stdlib 3.16.1) proc_lib.erl:226: :proc_lib.init_p_do_apply/3
```

fhunleth, 2021-10-31:
Hi @CharlesOkwuagwu! 

Regarding whether this was intentional or not, I don't recall having `open` take so long before either working or returning an error. Since I don't work hardly at all on Windows these days, it might be a new behavior with dealing with some serial ports.

In any case, I certainly wouldn't oppose having a settable timeout on `open`. I just wonder if something else is happening and perhaps an error should be detected and returned earlier. Do you have any idea why the `open` is taking so long on your system?

CharlesOkwuagwu, 2021-10-31:
I see this happen mainly on application restarts when maybe the modem had been in use prior and is just re-initializing itself.

What I then do is delay 30 seconds then retry the open, which always works.

closed 2021-10-31 by CharlesOkwuagwu