elixir-circuits/circuits_uart #114: Circuits.UART.enumerate no work on Ubuntu 20.04

braulio, 2021-11-06:
Hello.

Hello
I created a small application to test Circuits.UART. But I cannot enumerate the ports. I use iex -S mix and Circuits.UART.enumerate.

The result is always %{}

Need help.

Thanks.

fhunleth, 2021-11-08:
Could you check that it's possible to use the ports outside of Elixir? Are you a member of the `dialout` group?

braulio, 2021-11-12:
Hello Frank.

I managed to send a test message from /dev/pts/2 to /dev/pts/3 after
setting permissions with chmod a+rw /dev/pts/2. But Circuits.UART.enumerate
still doesn't work.
Interactive Elixir (1.12.2) - press Ctrl+C to exit (type h() ENTER for help)
iex(1)> Circuits.UART.enumerate
%{}

I still don't know what's going on.

Thank you!

fhunleth, 2021-11-12:
`/dev/pts/2` and `/dev/pts/3` are not UARTs, so it's expected that `Circuits.UART.enumerate` won't return them.