elixir-circuits/circuits_uart #117: Connection opens without errors, but I can't neither read or write

FabioFusimoto, 2022-01-05:
## Setup
  - Version: 1.4.3
  - OS: Windows 10
  - Platform: Desktop | Attempting to connect to an ESP32 Dev Module
  
## Expected Behavior
It should be possible to both send and receive messages through the serial port after the connection is opened.

## Actual Behavior
Calling `Circuits.UART.write` has no effect, despite it not rendering an error and  `Circuits.UART.read` always results in a timeout

## Steps to Reproduce the Problem
I'm using Circuits UART on a [personal project](https://github.com/FabioFusimoto/panacea/blob/main/lib/panacea/serial.ex). 

The ESP32 board is recognized properly by the `enumerate` function, which I use in order to decide which port to use. After opening the serial connection (which BTW returns `:ok` as expected), I try to call either `write` or `read` and have no success whatsoever while trying to communicate with and ESP32 Dev Kit running at 921600 bps baud rate.

The serial communication works properly if I run my project under Pop_OS! instead of Windows 10. Prior to using and ESP32, I was using an Arduino Uno board; which worked flawlessly on both Linux and Windows.

I've already tried to recompile the depencies by running `mix clean`, `mix compile` and `mix deps.compile` but the result remains the same.

Any idea on what I could try next?

fhunleth, 2022-01-08:
I'm sorry, I'm not sure since I haven't used 921600bps on Windows. My first try would be to lower the baud rate to 112500 to see if that works. It could be that something is not right with high baud rate support.

FabioFusimoto, 2022-01-10:
I've tried 115200, but still not working properly. I find it weird that normally the serial bus gets "busy" when a connection is properly established: meaning that attempting to upload a new program through Arduino's IDE or connecting through putty results in an error (something that indicates that said port is "already in use"). When using my ESP32 under Windows, it doesn't reach said "busy" state, since both uploading a new program and connecting via an external software remain possible.