elixir-circuits/circuits_uart #118: Circuits.UART.enumerate returns nothing %{"ttyAMA0" => %{}, "ttyS0" => %{}}

shahryarjb, 2022-01-09:
## Setup

  - Version: (if not latest, please try upgrading) `{:circuits_uart, "~> 1.4"}`
  - OS: OSX
  - Platform: Raspberry Pi 4
  
## Expected Behavior
I install circuits_uart on my Nerves project and after burning on it. I connect to my nerves and run this function: `Circuits.UART.enumerate`
[image]

```elixir
iex(5)> Circuits.UART.enumerate
%{"ttyAMA0" => %{}, "ttyS0" => %{}}
iex(6)> Circuits.UART.enumerate
%{"ttyAMA0" => %{}, "ttyS0" => %{}}
iex(7)> {:ok, pid} = Circuits.UART.start_link
{:ok, #PID<0.1130.0>}
iex(8)> Circuits.UART.enumerate
%{"ttyAMA0" => %{}, "ttyS0" => %{}}
```
but it can not find anything, although I connected 3 different USB to my Raspberry Pi 4:
[image]

Thank you

fhunleth, 2022-01-09:
Nerves doesn't ship with the device drivers for the USB serial cables that you're using. It has drivers for cables with these chips in them: https://github.com/nerves-project/nerves_system_rpi4/blob/main/linux-5.10.defconfig#L466-L469.

`"ttyS0"` goes to the GPIO header's UART pins. `"ttyAMA0"` goes to the Bluetooth module on the board. It looks like `"ttyS0"` is working for you. You specified active mode, so everything sent over the serial port will be turned into an Elixir message. Run `flush` to get those. `UART.read/2` returns an error since you specified active mode.

shahryarjb, 2022-01-09:
Thank you, @fhunleth, it is hard to understand for me, and I am very sorry I am trying to figure out how it works.

```
iex(1)> {:ok, uart} = Circuits.UART.start_link
{:ok, #PID<0.1124.0>}
iex(2)> Circuits.UART.open(uart, "ttyS0", speed: 9600,active: true)
:ok
iex(3)> Circuits.UART.write(uart, "Hello there\r\n")
:ok
iex(4)> flush
:ok
iex(5)> Circuits.UART.read(uart, 60000)
{:error, :einval}
```
I saw many topics and when they run `flush` they could see the outputs, but I just got a `:ok` atom. I just need something to test and see result to continue with what I receive. Furthermore, I changed `active: true` to `flase` and run `Circuits.UART.read(uart, 60000)` it took a 2 or 3 min to show `{:ok, ""}`

Again I am very sorry, I just want to create an open-source project for analysing cars Which is based on a thesis.
Thank you

fhunleth, 2022-01-09:
Nothing was received on the UART. I'm not sure what you're expecting to be received, so maybe it's working just fine?

fhunleth, 2022-01-09:
`Circuits.UART` also works on MacOS, so you could try it out there as well.

shahryarjb, 2022-01-09:
@fhunleth would you mind suggesting me a device or something exists in home to test its USB? I do not know how to send a file or something to my Raspberry Pi and get this with your library, it shows me `{:ok, ""}` because no device is connected to it.

If I find a device and connect with its USB, I think I can a way to connect to a car. I tried my mobile, but I didn't know how to send  a file or receive it.
@fhunleth would you mind suggesting me a device or something exists in home to test its USB? I do not know how to send a file or something to my Raspberry Pi and get this with your library.

If I find a device and connect with its USB, I think I can a way to connect to a car. I tried my mobile, but I didn't know how to send  a file or receive it.

fhunleth, 2022-01-09:
Can you connect a wire from the UART's TX pin to the UART's RX pin? Then you can send strings to yourself.

shahryarjb, 2022-01-09:
I think I can't, I do not know event these name ;), but I relay want to say thank you for all your efforts.
I should find a device to connect.

I have a question, when we want to write or receive, we should know some specialized String like `AT, AC` to connect a device, for example every device needs? And every company has the own String or Integer?

### For example:

If I run `Circuits.UART.write(uart, "AT")`, It gives me back the number of passengers, for example?!! If yes, what should I search about it? What is this keyword?

fhunleth, 2022-01-09:
I'm going to close this.

Before you ask more questions here, could you spend some time reading up on serial ports and serial communication? I think this will help quite a bit.

closed 2022-01-09 by fhunleth

shahryarjb, 2022-01-09:
@fhunleth, yes sure. May you suggest me a reference?

Before taking your time, I want to prepare a car ECU and `OBD` cable and test them, because I think I can't test it with my mobile or some devices exist in my home.

Thank you very much