elixir-circuits/circuits_uart #142: Error compiling with BeagleBone based target

patatoid, 2022-09-23:
## Setup

  - Version: 1.5.0
  - OS: Linux
  - Platform: Custom
  
## Expected Behavior

I am looking forward to compiling circuits UART targeting a custom device. We have built a system based on BeagleBone (https://github.com/nerves-project/nerves_system_bbb) but compilation fails.

## Actual Behavior

I am facing an issue compiling circuits_uart:

```
src/uart_comm_unix.c: In function 'uart_get_rs485_config':
src/uart_comm_unix.c:383:50: error: 'SER_RS485_TERMINATE_BUS' undeclared (first use in this function)
  383 |     config->rs485_terminate_bus = (rs485.flags & SER_RS485_TERMINATE_BUS) != 0;
      |                                                  ^~~~~~~~~~~~~~~~~~~~~~~
src/uart_comm_unix.c:383:50: note: each undeclared identifier is reported only once for each function it appears in
src/uart_comm_unix.c: In function 'uart_config_rs485':
src/uart_comm_unix.c:434:32: error: 'SER_RS485_TERMINATE_BUS' undeclared (first use in this function)
  434 |     update_flags(&rs485.flags, SER_RS485_TERMINATE_BUS, config->rs485_terminate_bus);
      |                                ^~~~~~~~~~~~~~~~~~~~~~~
make: *** [src/Makefile:107: /<***>/lib/circuits_uart/obj/uart_comm_unix.o] Error 1
```

## Steps to Reproduce the Problem

Compile circuits_uart with a BeagleBone-based system as a target should trigger the above error.

After some investigation, I saw that there were kernel headers missing in our target system, the whole uapi-related headers that include the required `SER_RS485_TERMINATE_BUS` constant.

fhunleth, 2022-09-23:
Hi @patatoid,

I just tried this by building [Circuits Quickstart](https://github.com/elixir-circuits/circuits_quickstart) using the Beaglebone target and wasn't able to reproduce. 

The Linux headers originate with the gcc toolchain used to build the system. In your custom system, did you change this? For example, for a BeagleBone system, I'd expect https://github.com/nerves-project/toolchains/releases/download/v1.6.0/nerves_toolchain_armv7_nerves_linux_gnueabihf-linux_x86_64-1.6.0-DAFF8EF.tar.xz to be used now.

patatoid, 2022-09-23:
I did not understand that the headers were to be present in the toolchain, I am still young with those concerns.

We were using the toolchain you mentioned but in an older version (the 1.4.3), I overrode the dependency to use the toolchain updated to the version you gave (1.6.0), but the headers to define the `SER_RS485_TERMINATE_BUS` are still not present. and failing with the same error.

fhunleth, 2022-09-23:
Could you look at line 119 of `~.nerves/artifacts/nerves_toolchain_armv7_nerves_linux_gnueabihf-linux_x86_64-1.6.0/armv7-nerves-linux-gnueabihf/sysroot/usr/include/linux/serial.h`?

I think something else is going on. Could you `export V=1` and rebuild? This should print out the commandline options that were passed to `gcc`. Could you also show how you're building your project? (i.e., are you running `mix firmware`?)

patatoid, 2022-09-23:
Here is the gcc command output:
```
/home/pascal/.nerves/artifacts/nerves_toolchain_armv7_nerves_linux_gnueabihf-linux_x86_64-1.6.0/bin/armv7-nerves-linux-gnueabihf-gcc -c -I/home/pascal/.nerves/artifacts/nerves_system_c2-portable-0.3.0/staging/usr/lib/erlang/erts-11.1.7/include -I/home/pascal/.nerves/artifacts/nerves_system_c2-portable-0.3.0/staging/usr/lib/erlang/lib/erl_interface-4.0.2/include -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64  -pipe -O2 -I/home/pascal/.nerves/artifacts/nerves_system_c2-portable-0.3.0/staging/usr/include -std=c99 -D_GNU_SOURCE -o /home/pascal/malach.it/zola/aloha/_build/c2_dev/lib/circuits_uart/obj/uart_comm_unix.o src/uart_comm_unix.c
```

To build the project, I use the command `mix firmware`.

jeanparpaillon, 2022-10-03:
@fhunleth I've taken over this task in our organization. fyi, we are using an older kernel (4.19) where `SER_RS485_TERMINATE_BUS` constant was missing (maybe due to configuration, because it was introduced in 4.12-rc1)
I think we can close this, as this was introduced in a 5 years old kernel that may be considered as obsolete for Nerves users.
@patatoid wdyt ?

fhunleth, 2022-10-03:
Ahh. I see. It's certainly ok with me to make RS486 support conditional if you'd like to send a PR.

patatoid, 2022-10-04:
@jeanparpaillon definitely it is the `circuits_uart` update from `1.4.3` to `1.5.0` that introduced `RS485` which needed a newer kernel as stated above.

closed 2022-10-04 by patatoid